puuu/USIWire #2: test failure

trombik, 2017-07-16:
hi,

I am trying the provided tests but failed to reproduce reliable test results.

the setup:

* arduino nano as master
* attiny85 as slave
* both are on a breadboard
* the 5V power is provided by USB cable to nano
* no other discrete parts

without modification to the example code, the test keeps failing. I guessed failures were caused by timing, and inserted some `delay()`. after trials and errors, sometimes, the test passes, but not always. do you see anything I am missing? `AUX_Callback` test fails in most cases, but now always. here is the diff, failed result, and successful one.

```diff
--- ../src/TestMaster/TestMaster.ino	2017-07-16 23:03:06.650753365 +0900
+++ src/TestMaster.ino	2017-07-17 00:29:23.520245148 +0900
@@ -611,7 +611,7 @@
     Wire.write(CONTROL_ADDR);
     Wire.write(AUX_STATE_ON << CONTROL_AUX_POS | slaveSleepMode);
     assertEqual(Wire.endTransmission(), 0);
-    delay(1);
+    delay(20);
     if (slaveSleepMode) delay(2);
     assertEqual(digitalRead(AUX_PIN), HIGH);
     // AUX pin low
@@ -619,7 +619,7 @@
     Wire.write(CONTROL_ADDR);
     Wire.write(AUX_STATE_OFF << CONTROL_AUX_POS | slaveSleepMode);
     assertEqual(Wire.endTransmission(), 0);
-    delay(1);
+    delay(20);
     if (slaveSleepMode) delay(2);
     assertEqual(digitalRead(AUX_PIN), LOW);
   }
@@ -639,8 +639,8 @@
       Wire.write(i);
     }
     assertEqual(Wire.endTransmission(), 3);
-    delay(1);
-    if (slaveSleepMode) delay(2);
+    delay(10);
+    if (slaveSleepMode) delay(20);
     assertEqual(digitalRead(AUX_PIN), HIGH);
     // AUX pin low
     Wire.beginTransmission(SLAVE_ADDR);
@@ -650,8 +650,8 @@
       Wire.write(i);
     }
     assertEqual(Wire.endTransmission(), 3);
-    delay(1);
-    if (slaveSleepMode) delay(2);
+    delay(10);
+    if (slaveSleepMode) delay(20);
     assertEqual(digitalRead(AUX_PIN), LOW);
   }
 }
@@ -672,8 +672,9 @@
     assertEqual(Wire.endTransmission(false), 0); // send no Stop Condition
     assertEqual(digitalRead(AUX_PIN), LOW); // still low
     Wire.beginTransmission(SLAVE_ADDR);
+    delay(10);
     assertEqual(Wire.endTransmission(false), 0); // repeat start
-    delay(1);
+    delay(10);
     if (slaveSleepMode) delay(2);
     assertEqual(digitalRead(AUX_PIN), HIGH);
     // request AUX pin low
@@ -684,7 +685,7 @@
     assertEqual(digitalRead(AUX_PIN), HIGH); // still high
[... 224 lines cut ...]
SMBus_Quick_Write_Command: . SUCCESS!
SMBus_Receive_Byte: .... SUCCESS!
SMBus_Send_Byte: . SUCCESS!
SMBus_Read_Byte: ... SUCCESS!
SMBus_Read_Word: .... SUCCESS!
SMBus_Write_Byte: .... SUCCESS!
SMBus_Write_Word: ..... SUCCESS!
SMBus_Process_Call: ........ SUCCESS!
SMBus_Block_Read: ................. SUCCESS!
SMBus_Block_Write: ................. SUCCESS!
SMBus_Block_Process_Call: ........................ SUCCESS!
SMBus_Quick_Read_Command_Wrong_Addr: ..... SUCCESS!
SMBus_Quick_Write_Command_Wrong_Addr: ..... SUCCESS!
SMBus_Read_Byte_Multiple_Times: .................. SUCCESS!
Read_Too_Much: .... SUCCESS!
Read_Too_Less: ...................... SUCCESS!
Write_Too_Much: ............ SUCCESS!
Write_Too_Less: .......... SUCCESS!
AUX_On_Off_Repeated_Start: ........................................... SUCCESS!
AUX_Callback: ................... SUCCESS!
Fill_Slave_RX_Buffer: ...................... SUCCESS!
AUX_On_Off_Repeated_Start_NACK: ........................................... SUCCESS!
AUX_Callback_NACK: .................. SUCCESS!
All test finished!
```

puuu, 2017-07-26:
Hi,
thank you for testing USIWire.

Unfortunately, I can confirm that the test not always run successfully. Tuning of the `delay()` may help. The current `delay()` values were tested with a ATTiny85 at 8 MHz. With these I can run at least 8 test in a row without any failure.

What clock you are using?

trombik, 2017-07-28:
I think it is 8 Mhz. but will double-check.

it is very helpful for a library to have tests (most of them do not), but it would be nicer if the test configuration was documented, such as the chips used as master and slave in the test, wiring, breadboard or not, etc.