puuu/USIWire #4: Library Clarification

frozenspongebob, 2017-09-23:
Hi,

This is not so much of an issue than it is a question for clarification. If I understand the purpose of this library correct, this library utilizes Universal Serial Interface (USI) to provide I2C Master and Slave functionality for the ATtiny family (those having USI capabilities), correct? In other words, when programming an ATtiny (say, ATtiny24a, for instance) using Arduino as an ISP, USIWire will allow the **ATtiny** (disconnected from the Arduino and running standalone) to act as a master and communicate with a slave, correct?

Also, is `#ifdef ESP8266  // default value of 230 us is too short
  Wire.setClockStretchLimit(1500);
#endif` necessary for communication with ANY slave?

Thanks so much!
Jonathan Lee
bukatea@gmail.com

liebman, 2017-09-23:
When using an ESP8266 as a master with ATtiny slave,  I found that if your running the ATtiny at 8mhz or above its not needed.  Even 4mhz I found it "mostly" works.  At 1mhz its absolutely required as the slave will do long clock stretches and the ESP8266 will timeout, sometimes leaving the i2c buss in a bad state.

frozenspongebob, 2017-09-23:
Sorry if I wasn't clear. I mean using the ATtiny as a master to control a slave, such as a battery fuel gauge, not ESP8266 as a master of the ATtiny slave. is that statement necessary?

liebman, 2017-09-23:
That statement is only needed for an ESP8266 master. (That's why it's #ifdef'd)

frozenspongebob, 2017-09-29:
Thanks for clarifying. To be sure, this library allows the user to use the Wire library from Arduino when you're programming an ATtiny (say, 44a) using Arduino's ISP Programmer capabilities, right? In other words, can you upload a piece of code with a command such as `Wire.beginTransmission(ADDRESS_HERE);` to the ATtiny using the Arduino ISP sketch?

Thanks so much!
Jonathan Lee
bukatea@gmail.com

liebman, 2017-09-29:
Yes, while I've only used this as a slave, master is fully implemented.

frozenspongebob, 2017-09-29:
Thanks so much! You've been a great help.

closed 2017-09-29 by frozenspongebob