puuu/USIWire #8: I2C speed for 1Mhz internal oscillator

dontsovcmc, 2018-04-11:
Do you know, what i2c speed is able to use with Attiny85 (1Mhz internal oscillator)
Here is some text with 33kHz limit: https://github.com/felias-fogg/SoftI2CMaster
Can you add change speed option? Thanks a lot!

puuu, 2018-04-12:
I never tried the i2c speed with a ATtiny at 1 MHz. But I expect more than 33kHz with this USI implementation. Also USI supports clock stretching. Please feel free to test and report!

dontsovcmc, 2018-04-12:
Thanks! How can I change speed in your library (i2c slave)?

liebman, 2018-04-12:
I run i2c client on an ATtiny85 (1mhz internal oscillator) without issues in this project [AnalogClock](https://github.com/liebman/AnalogClock) (See the I2CAnalogClock sub-directory) without issues.

dontsovcmc, 2018-04-12:
See the trace. It works for short messages:
2018-04-11 02:56:58,787 - root - INFO - handle_data (134): 78 00 01 40 1e 00 01 00 1c 05 eb 2c 43 0a 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
2018-04-11 03:27:29,241 - root - INFO - handle_data (134): 78 00 01 40 1e 00 01 00 1c 0b eb 2c 43 0a 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 01 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

puuu, 2018-04-12:
@liebman : thank you for reporting!

@dontsovcmc 
> How can I change speed in your library (i2c slave)?

By i2c, the speed is given by the master. There is no option to change the speed on the slave side.

dontsovcmc, 2018-04-12:
@liebman I don't use SoftI2CMaster. I use USIWire (slave) and Wire (master). I try to find any information about i2c speed and found text about 33khz.

@puuu wow.. good news :). I'll test it.

dontsovcmc, 2018-04-14:
I change speed.
Wire.setClock( 33000L );
```
2018-04-13 23:18:03,729 - root - INFO - handle_data (134): 78 00 01 01 1e 00 01 00 92 06 eb 2c 43 0a 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 00 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 ff ff ff ff ff ff ff ff ff
2018-04-13 23:48:41,470 - root - INFO - handle_data (134): 78 00 01 01 1e 00 01 00 92 0c eb 2c 43 0a 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 00 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

puuu, 2018-04-14:
@dontsovcmc I do not know what your output comes from neither I know what should be the expected output.

Please, as requested by @liebman, describe your set-up, i2c slave, i2c master, other i2c devices on the bus and their connection and used i2c implementations. If you have any issues, please provide a reduces example code and try at different CPU clock speeds.

Further on, `Wire.setClock()` is for [i2c master](https://www.arduino.cc/en/Reference/WireSetClock) and for USIWire it is even [not implemented](https://github.com/puuu/USIWire/blob/master/src/USIWire.cpp#L70).

dontsovcmc, 2018-04-14:
Thanks a lot for the help! 

Attiny85 (Slave)
Library: https://github.com/puuu/USIWire/tree/master/src/USI_TWI_Slave
https://github.com/dontsovcmc/ImpCounter/blob/master/ImpCounter85/src/SlaveI2C.cpp

ESP-01 (Master)
Library: https://github.com/esp8266/Arduino/tree/master/libraries/Wire
Pin 0: SDA, pull-up 3k3
Pin 2: SCL, pull-up 3k3
Power: 3.2V
https://github.com/dontsovcmc/ImpCounter/blob/master/ImpCounterESP/src/MasterI2C.cpp#L33

Board: https://github.com/dontsovcmc/ImpCounter/blob/master/Board/board-ESP-01-1layer.png

dontsovcmc, 2018-04-15:
I got velleman pcsu200 oscilloscope:
https://www.velleman.eu/products/view/?id=407512
But can't understand how to record all messages =(. When I use trigger it stopped after event.
Record mode is very slow. 
This is [triggered first bytes](https://ibb.co/f1shsS). I reseted ESP (i2c master). The slave is sleeping. So, I got only FF FF FF.

dontsovcmc, 2018-04-15:
Wow! There are a lot of problems between ESP8266 library & slow i2c devices.
- First of all: Wire supported  [only 50khz between 100](https://github.com/esp8266/Arduino/blob/2105b8b06fef3e155dcdb775436ca1f77dccd586/cores/esp8266/core_esp8266_si2c.c#L47)
- i2c libraries are platform dependence.

My debug roadmap:
2. Try [setClockStretchLimit](https://github.com/esp8266/Arduino/issues/3977)() function. Default: 230. In some cases uses increase it to 800-2000.
3. some [repair](https://github.com/enjoyneering/ESP8266-I2C-Driver). check it.
4. fast i2c on assempler. ttps://github.com/pasko-zh/brzo_i2c
5. slow i2c support: https://github.com/Seeed-Studio/Arduino_Software_I2C
6. also https://github.com/felias-fogg/SlowSoftI2CMaster

liebman, 2018-04-16:
@dontsovcmc The project I refereed to earlier uses an esp8266 master with a ds3231 RTC and an ATTiny85 as slaves with the tiny running at 1mhz.  Tho only initialization I do for Wire is to set the clock stretch limit. Leaving the speed the default 100khz.

```
Wire.begin();
Wire.setClockStretchLimit(1500);
```
Do be sure to check the return value from wire commands as many of those can detect errors.

dontsovcmc, 2018-04-17:
Fantastic! Thanks a lot @liebman!
I set setClockStretchLimit(1500) and there is no any errors during 40 times. 130 bytes received correctly. 100khz speed. 

So, for communication with slow devices, like Attiny85 1mhz internal,  we should change i2c master stretch limit : setClockStretchLimit to ~1500.

closed 2018-04-17 by dontsovcmc

dontsovcmc, 2018-04-18:
@liebman oh no... 80 times was perfect, but later.. i2c failed =(. Attiny85 wake up ESP, but don't answer commands.