puuu/USIWire #11: USIWire compile error

adrianmihalko, 2020-03-21:
Is there any compatibility issue with this library and Arduino 1.8.12?

Without Usiwire library I get error about missing library (as expected)

```
In file included from /Users/adrianmihalko/Downloads/attiny_daemon-master/firmware/ATTinyDaemon/ATTinyDaemon.ino:1:0:
ATTinyDaemon.h:1:10: fatal error: USIWire.h: No such file or directory
 #include <USIWire.h>
          ^~~~~~~~~~~
compilation terminated.
exit status 1
USIWire.h: No such file or directory
Now, I install library as Sketch - Include Library - Add .ZIP Library and I choose USIWire-0.1.0.zip.

```
I click Very/Compile and I get the error:

```
In file included from /Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USI_TWI_Master/USI_TWI_Master.h:76:0,
                 from /Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USIWire.cpp:27:
/Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USI_TWI_Master/usi_io.h:1:1: error: expected unqualified-id before '.' token
 ../USI_TWI_Slave/usi_io.h
 ^
/Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USIWire.cpp: In member function 'void USIWire::begin()':
/Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USIWire.cpp:55:3: error: 'USI_TWI_Master_Initialise' was not declared in this scope
   USI_TWI_Master_Initialise();
   ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USIWire.cpp:55:3: note: suggested alternative: 'USI_TWI_Slave_Initialise'
   USI_TWI_Master_Initialise();
   ^~~~~~~~~~~~~~~~~~~~~~~~~
   USI_TWI_Slave_Initialise
exit status 1
Error compiling for board ATtiny25/45/85.
```

I am not alone with this problem: 

https://github.com/jbaumann/attiny_daemon/issues/7

puuu, 2020-03-22:
Please show a minimal example and describe the settings and the version you are using (ATTiny Core).

I cannot reproduce that. The examples `TestMaster` and `TestSlave` compile for me. Tested with:
- Arduino:  1.8.12
- ATTinyCore: 1.3.3
- Board: ATtiny25/45/85
- Chip: ATtiny85

jbaumann, 2020-03-23:
[copied from original issue]
Ok, I'm not sure how this happens, but if you use "Sketch - Include Library - Add .ZIP Library", then the file contains only the following: "../USI_TWI_Slave/usi_io.h". This is clearly wrong and two possible explanations are on the top of my mind:
- The original author uses a symbolic link that is interpreted wrongly by the Arduino IDE when unpacking the files
- The Zip-Archive knows that both files are the same and saves only a reference, which again is interpreted wrongly by the Arduino IDE
In both case I would see the problem in the Arduino IDE.

Two simple possible fixes:
- Copy the file from the TWI_Slave over to the TWI_Master
- Remove the directory, unpack the library manually and copy it to your libraries folder

I suggest the second way as the general way to install libraries. This ensures that everything that goes wrong comes from your own errors.

adrianmihalko, 2020-03-24:
Release 0.2.0 solved the issue.

closed 2020-03-24 by adrianmihalko

puuu, 2020-03-24:
Thank you for reporting and pointing the issue. Version 0.2.0 was released, which no longer contains symbolic links.

According to the [Library Manager FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ), symlinks are no longer allowed, this was documented in February 2018. This is probably also the reason why USIWrire disappeared from the Arduino Library Manager.