puuu/USIWire #17: Bundling USIWire with MegaCore

MCUdude, 2022-04-26:
Hi!

I'm currently working on adding Arduino support for ATmega165/325/645 and ATmega169/329/649 through an Arduino core called [MegaCore](https://github.com/MCUdude/MegaCore/tree/atmegaxx5-atmegaxx9-support). The catch is that they don't have a "standard" TWI interface, but USI instead.

I was wondering. Is it OK for you if I bundle a modified version of this library together with MegaCore? I'll of course give you the credit for your excellent work. The only modifications I'm planning to use is to change the class name from `USIWire` to `TwoWire` to make it compatible with other libraries that expect a `TwoWire` object. I'll also add a little preprocessor glue so that users can include Wire.h, and it automatically includes the USIWire library.

Thanks!

puuu, 2022-04-27:
Hi, nice to hear that USIWire is of interest.

Feel free to use USIWire as you like. In fact, [ATTinyCore](/SpenceKonde/ATTinyCore) bundled it. Unfortunately, changes do not come back to this repository. Please also check ATTinyCore for modifications.

Why `TwoWire`? isn't it just called `Wire` in Arduino?

IMHO, the cleanest way would be to use this repository as a git submodule. The glue would be a `Wire.h` (in a different directory) with `include <USIWire.h>`. [USIWire.h](https://github.com/puuu/USIWire/blob/master/src/USIWire.h#L75) introduce already `Wire`. If you need `TwoWire` add a additional line into your `Wire.h`.

MCUdude, 2022-04-27:
> Feel free to use USIWire as you like. In fact, [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore) bundled it. Unfortunately, changes do not come back to this repository. Please also check ATTinyCore for modifications.

Thanks! I'll have a look at both libraries then.

> Why TwoWire? isn't it just called Wire in Arduino?

The library itself and the object is called Wire. However, the class is called TwoWire. [Here](https://github.com/MCUdude/SigmaDSP/blob/f2a7be955a124bddbd94d2342afc2514b9276a07/src/SigmaDSP.cpp#L16-L20) and [here](https://github.com/adafruit/Adafruit_SSD1306/blob/673fc5d974eec96d6368c46d48a81e9447463b61/Adafruit_SSD1306.cpp#L171-L181) is examples of libraries that expect a TwoWire object to be passed to their constructors as arguments. These libraries wouldn't accept USIWire as their i2c driver.

I'm using submodules for the bootloader and Arduino core files, but I would prefer to have a copy of the library that I'm free to modify the code as I like. One example is changing the class name (USIWire -> TwoWire). Another example is to get this library to compile for the targets I'm using. At the moment this library does not compile for `__AVR_ATmega325P__` and `__AVR_ATmega645P__`, so that's something I'll have to add to usi_io.h. The code examples bundled with this library appear to be written with ATtinys in mind, so the examples would also have to modified.

puuu, 2022-04-27:
> The library itself and the object is called Wire. However, the class is called TwoWire. [...]

I see, `TwoWire` is the class name. In this case `typedef USIWire TwoWire;` or more C++ style `using TwoWire = USIWire;` would do the trick in `Wire.h`.

> At the moment this library does not compile for `__AVR_ATmega325P__` and `__AVR_ATmega645P__`, so that's something I'll have to add to usi_io.h.

I would be happy about such *tested* pull requests.